BenningtonCS/Telescope-2014 #111: Dongle doesn't work after being plugged in for too long

hcrowl, 2015-06-18:
The dongle seems to only work at an "ideal" temperature. The performance is very unreliable if the dongle has been plugged in for too little time or too much time. We should both quantify the time over which the dongle is reliable and look to find a less flakey solution.

theDarkLard, 2015-06-19:
Test in progress. Dongle reached stable operating temperature @ 10:46 am. I am checking to see if we still get a signal about every half hour. It's 12:15 pm now and so far so good.

theDarkLard, 2015-06-19:
4:10 pm and the dongle still seems to be working, although the signal isn't quite as strong as it been before, but that could be due to a host of other issues.

theDarkLard, 2015-06-22:
After being unplugged all weekend, plugged dongle in @ around 9:50 am today, was working at that time. It is now about 11:40 am, and am getting a flat response. Unplugged it for a minute and plugged it back in, is wokring again now.  It seems the dongle is a little more volatile that I had suspected. Will continue to search for dongle replacements.

theDarkLard, 2015-06-22:
Leaving the dongle plugged in for now, will check back in an hour.

hcrowl, 2015-06-22:
A theory worth checking: it's not the dongle, but rather the computer going into standby and not recovering properly to sense the dongle. No idea if this is true, but it's worth testing to see if you can get it to fail just by having the computer going into power saving mode.

theDarkLard, 2015-06-22:
Just woke up the computer from power saving mode and immediately initiated srtn. The dongle still appears to be working. Will try it again the next time I can get away from the stockroom computer for a bit.

edaniszewski, 2015-06-22:
#44 perhaps related?

theDarkLard, 2015-06-22:
@edaniszewski I don't think so because that issue happens only when srtn is already running and the computer goes to power saving mode. In this case, the dongle will have been plugged in with srtn not running and you run srtn and you get a flatline.

4:00 pm and dongle is still working fine. Go figure.

edaniszewski, 2015-06-22:
Interesting. Seems like a tricky one to diagnose (if it indeed is not a problem with the dongle).

I don't really remember the ins-and-outs of interfacing with the dongle, but I'm sure there has to be a way to extract the data from the dongle external from the srtn software. Isolating the dongle data stream from the software will give you one (or a few) less degree of uncertainty since it is plausible that the software is a bit wonky?

theDarkLard, 2015-06-23:
Cool yeah that sounds like a good diagnostic check. I know there is the driver for the dongle in the srtnver3 directory but it's pretty messy. I will try to look into getting at the dongle without going through srtn but I can't promise I will come up with anything.

edaniszewski, 2015-06-23:
since srtn does access the dongle via the driver, pretty much all the code should be there.. somewhere. If we can pinpoint exactly which code is accessing the driver, we could just use it in a small standalone program to log the data coming from it.

The secret to writing code: copy + paste.

hcrowl, 2015-07-07:
Any progress on this? Is it true that this hasn't crashed since we took the case off of the dongle?

edaniszewski, 2015-07-07:
If having a diagnostic check independent from the srt software is still desirable, there is a python wrapper for the C API
https://github.com/roger-/pyrtlsdr

should be easy enough to write a small python script to collect data throughout the day.

theDarkLard, 2015-07-07:
Yeah, @hcrowl, no crashes since you took the casing off.

theDarkLard, 2015-07-07:
I would say it's fine to close this.

closed 2015-07-07 by theDarkLard