BenningtonCS/Telescope-2014 #112: Calibration procedure

hcrowl, 2015-06-18:
Figure out how to "calibrate" the radio telescope. What does the "Calibrate" button work? How do we use it?

theDarkLard, 2015-06-19:
So I ended up trying the procedure they list out for the old srtn software, the one written in java, and i think it may have worked..? Here's the procedure and what happened: point the telescope at an empty point in the sky. Switch the frequency to just off the hydrogen line, to 1419 MHz, hit the cal button. Once the green light goes on, switch the frequency back the 1420.4 MHz. This stops the cal button from flashing green. When I went back to 1420.4 MHz the signal I got from the sun was too large to be displayed, it just hit the top of the display window, but this probably would have been adjusted on the wrap around.

theDarkLard, 2015-06-19:
This does indeed get fixed on the wrap around. The signal strength is still slowly ascending but it is not as drastic as it was yesterday, and it was doing that before the calibration, so I assume it has something to do with how close the telescope is to the sun's center.

hcrowl, 2015-06-19:
Yeah. The thing to remember is that the telescope isn't tracking, so it's
going to slowly scan across the Sun. When it's tracking away from the Sun's
center, it should slowly decline.

Not saying that's necessarily what's happening here, but that's a
possibility.

theDarkLard, 2015-06-19:
Actually when you click on the sun from the skymap, the telescope does indeed track it. So if this weird ascending is a consequence of the tracking, i think the telescope is sort of anticipating the sun, where it is moved a little in front of it and sits there as the sun's center aligns more closely with the center of the beam. But again, this is all speculation. As far as recording data goes, you hit the record button and everything gets dumped into a text file. you have to then extract it and make a plot in MS Excel or something similar.

edaniszewski, 2015-06-19:
> you have to then extract it and make a plot in MS Excel or something similar.

@theDarkLard It would be easy enough to automate this using either thin python layer (matplotlib?), or adding some functionality to the C code (not sure what plotting libraries exist, but I'm sure something is out there.)

theDarkLard, 2015-06-19:
Hey, we were hoping you'd show up!

edaniszewski, 2015-06-21:
Big brother is _always_ watching (;

theDarkLard, 2015-06-23:
So yeah yesterday I put the calibration procedure in the srtn help file on the wiki and you literally point at an absorber or tree and click. At least, that's all the help file lets on. But I think we should probably get an absorber of some kind, because there is a LOT of noise up on the roof. The signal on the horizon vs on cold sky is gigantic, so I'm going to try to characterize the signal strength all around the horizon to see if it's all bad or if there is a lone tree that we could use. I say this because when I point the telescope at the most obvious tree, it gets the same signal strength it gets in its stow position, which is pointing almost directly at the big AC unit.

edaniszewski, 2015-06-23:
When pointing at the most obvious tree, is the big mountain in line with the telescope? I vaguely remember us being concerned that there may be interference from there, since there is a bunch of telecom equipment up there?

theDarkLard, 2015-06-23:
no it's sort of pointed towards the new houses/stickney

hcrowl, 2015-06-23:
Maybe we should use the spectrum analyzer to characterize noise sources on the roof?

theDarkLard, 2015-06-23:
yeah good idea^^ will do that

edaniszewski, 2015-06-23:
ohhh! toys!

You should livesteam it, so I can have fun and pretend like I'm there

theDarkLard, 2015-06-23:
you can! once I get the camera fully working...... :/

theDarkLard, 2015-06-23:
Ok so just went up on the roof w/ the spectrum analyzer. Yes there is a good deal of noise all around the horizon but it is all > 1 GHz. Strongest signal was -50 dBm at 93 MHz. Other prominent frequencies were 180 MHz, 201 MHz, and 663/666 MHz. The 93 and 201 MHz signals were the most pervasive, showing up everywhere. I don't know how much this noise is going to affect our telescope. Theoretically it should all be filtered out.. there were no 1420 MHz signals around. I'm going to go point the telescope at some places around the horizon and try to find the best tree for calibration.

theDarkLard, 2015-06-23:
I hooked the spectrum analyzer back up to the telescope circuit, and the only frequency below 1 GHz that gets through is 249 MHz with a strength of 69-71 dBm, so this is the only real "noise" that's making it thorugh. The rest of the spectrum is in the 1420 MHz range, with comparable signal strength.

theDarkLard, 2015-06-23:
^^All of this on the horizon, pointing at different trees.

theDarkLard, 2015-06-23:
This makes me more confident in the accuracy of the hardware initial Tsys for our telescope. But these words could come back to haunt me...the radio telescope can be a harsh and cruel mistress, a fact we are all well aware of.

theDarkLard, 2015-06-23:
Still no idea on how to turn the calibration function off without changing the center frequency.

theDarkLard, 2015-06-23:
Looked at the sun with and without a calibration procedure done. Without, the power output is around 800-900. With calibration, there is virtually no difference in the power output. Still getting a steadily ascending power output. But the issue is that it seems like there is no real way to run the calibration procedure, turn it off, and maintain the calibrated spectrum/signal. Or at least if it's working, it's very hard to tell.

theDarkLard, 2015-06-23:
Roof noise stops affecting telescope at 30 degrees off the horizon. pwr when looking at cold sky is around 165, which would give a Tant of about 33 K.

theDarkLard, 2015-06-23:
Ok so I found something that could either be nothing or very troubling and very hilarious. So there are four (0, 1, 2, 3) calibration modes that the srtn calibration has. Mode 3 is used for tree absorbers. The funny thing is that there is no code written for mode 3 in cal.c. Mode 3 is the same as mode 2 but doesn't require an entry to confirm that you've placed an absrober in front of the feed. But I think this may have something to do with the cal button not turning off. I'm not going to post the entire cal.c file here but if you browse through it you will notice there is nothing written for "if mode = 3".

theDarkLard, 2015-06-23:
calmode 2 doesn't seem to want to turn off either.

edaniszewski, 2015-06-23:
Hm. Calmode 3 seems to only be mentioned in one place in the code, `disp.c` line 319

```C
if (d1.calmode == 3) {
        d1.docal = 1;
        return;
}
```

not too sure what the dealio is there, or why its only in disp.c... weird. Maybe I'll take a look at the MIT srtn ver4 sometime. Perhaps in the update from v3, they added some functionality around this? Im not too optimistic, but we'll see..

What exactly do you mean by "doesnt seem to want to turn off"? Its been a while since I've played around with the software, so I forget how it behaves. Is it that when you try and calibrate, there is no way to stop the calibration?

theDarkLard, 2015-06-24:
Yeah basically, you press the button and it stays green and doesn't go back to normal.

edaniszewski, 2015-06-24:
Interesting. Looking at the code, it looks like the the only place where the button is set to turn green is in `cal.c`. It seems like it should only be turning green if mode == 2

```C
if (d1.displ) {
    color.green = 0xffff;
    color.red = 0;
    color.blue = 0;
    gdk_color_parse("green", &color);
    gtk_widget_modify_bg(button_cal, GTK_STATE_NORMAL, &color);
}
```

the place where it appears to be set back to its default white color is in `disp.c`, though there is also a code block in `main.c` which assigns the button to white.

```C
if (d1.caldone) {
    d1.caldone = 0;
    color.green = 0xffff;
    color.red = 0xffff;
    color.blue = 0xffff;
    gtk_widget_modify_bg(button_cal, GTK_STATE_NORMAL, &color);
}
```

notice that is is checking for the `d1.caldone` flag, where caldone is probably set somewhere when the calibration is compete.

Hard to say whats going on. Doing a full code analysis would take a long time, so throwing out some guesses... it may just never be finished calibrating, it may be that it is never meant to turn back to white after the calibration is done, or it could just be something to do with the software hanging (though I think this last one is unlikely)

theDarkLard, 2015-06-24:
Hmmmm...thanks a ton for this Erick, this helps a lot. With this knowledge in mind, I'm going to see if anything more is illuminated by a few more practical tests. If not...well, we'll cross that bridge when we come to it.

theDarkLard, 2015-06-24:
So calmode 20 is actually the one for the absorber. It does not turn off either :( I do think though that the calibration process simply is not turning off. I ran the calibration and then pointed the telescope at a number of different points and the spectra kept changing all the while, seemingly incorporating the spectra/signal strengths of associated with each point into the continuing development of the av spectrum and discrete spectra.

theDarkLard, 2015-06-24:
Otherwise, as @edaniscewski posited, the procedure isn't meant to turn off or the button is meant to stay green. But I don't have a way of knowing for sure what is going on. Maybe we should pop off an email to Alan Rogers or to your friend Meredith @hcrowl?

theDarkLard, 2015-06-24:
I've also been trawling though the code some more but with no luck.

theDarkLard, 2015-06-24:
Oh and I should say a little bit about calmode 2. When you run the calibration, a little message bar pops up that says "enter to start bandpass cal,"  so I guess it must do something entirely different..? Calmode 20 displays "place absorber enter," so that's definitely the absorber mode. Obviously this is contradictory to what the help file says. Also, I've seen this before but hadn't considered it seriously until now, there is a sprintf command in cal.c that's supposed to print "calibration in progress." This has never come up in our version of srtn, at least not that I've noticed. It appears a bit after the line d1.caldone == 1, so it seems like the code is telling the calibration process to finish and then to print the in progress statement? This "calibration in progress" bit appears in cal.c. Sorry for all the thinking out loud and the ramblings of someone who doesn't know c...

hcrowl, 2015-06-24:
Here is a procedure derived by Meredith Hughes (Wesleyan University):

Intro level: bandpass only
Usually I'm just having them measure galactic rotation or the antenna beam width on the Sun, neither of which requires a precise Tsys calibration.  But galactic rotation does require a good bandpass calibration.  For this I just have them use the automated routine, which does bandpass beautifully (but not Tsys, as far as I can tell).  The procedure is:
1) Point the telescope at a patch of blank sky.  Usually about 45 degrees altitude in the north works pretty well. 
2) Hit the "cal" button on the software interface.  It will say "Place absorber enter:"
3) Place the absorber in front of the receiver.  I have an "absorber on a stick" that a student actually just goes out and holds in front of the receiver, making sure that as much of the antenna coil is covered as possible. 
4) When the absorber is in place, have another student hit "enter."  This will start the calibration.  As soon as it starts recording data, you're good to go.  If you're still pointed at blank sky (sometimes you have to move the antenna down to the horizon so that the student with the absorber can reach the receiver), the bandpass should now look flat, modulo some RFI. 

Advanced level: bandpass and Tsys
This is what I have students do if we actually care about measuring antenna temperatures -- for example, if we're trying to measure the brightness temperature of the Sun or Moon, or doing an aperture efficiency calculation on Cas A. 
1) First, do the bandpass calibration using the automated routine as described above. 
2) Then, move the antenna to a blank patch of sky, preferably at the altitude of the object you're planning to observe (for stability of measurements).  Record the power output (I think this number is displayed below the middle of the three spectral window plots).  This will be our "cold load" measurement P_cold, and we generally assume that blank sky is 3K.  (It is, of course, possible to do a sky dip and measure the atmospheric opacity to calculate the sky contribution, but I haven't found it necessary to bother.) 
3) Then, place the absorber in front of the receiver and record the total power measurement.  This will be our "hot load" measurement P_hot, and we generally either have a thermometer recording ambient temperature, or look up the current ambient temperature on the NOAA website. 
4) Once you have hot and cold load measurements, you can do a Y-factor calculation to get Tsys.  Specifically, if Y = P_hot / P_cold, then Tsys = (T_h - Y*T_c) / (Y-1), where T_h is the hot load temperature (the ambient temperature on the thermometer), and T_c is the cold load temperature (3K).  From this, you can also derive a gain factor to apply to any future power measurement to turn the power from the telescope into an antenna temperature.  For example, P_hot = G * (Tsys + T_h) allows you to calculate G, and then you can generalize so that P = G * (Tsys + Tant) to turn any power measurement P into an antenna temperature Tant.

hcrowl, 2015-06-24:
Oh! And the "absorber on a stick" that Wesleyan uses is a piece of "Eccosorb AN-77" on the end of an extendable rod.

edaniszewski, 2015-06-24:
It would be cool if we had a little device that could cover the receiver with an absorber with a push of a button. Maybe I dream too big though. 

Its good that we have a calibration process - some of it seems needlessly manual and could be/should be handled by the software. It is what it is, I guess.

theDarkLard, 2015-06-25:
I love how there's a fancy letter-digit name for a piece of foam.

edaniszewski, 2015-06-25:
It's designer foam :cocktail:

hcrowl, 2015-06-25:
It *is* pretty fancy foam. 

Having a push-button way to have the foam go into the beam would be cool. On the other hand, I'd be slightly worried about the foam obstructing something it shouldn't when it was in its "stowed" position. 

I agree that it's a bit manual. There's part of me that, from an educational perspective, like's the manual nature of it. On the other hand, some automation would be pleasing. This is maybe something that @vpascow should look at for his advanced work?

closed 2015-07-10 by theDarkLard