BenningtonCS/Telescope-2014 #114: Mount Hard Limits

hcrowl, 2015-06-18:
Figure out how to implement hard limits to protect the telescope. It should not allow anyone using the software to point the telescope anywhere that could damage the telescope.

theDarkLard, 2015-06-19:
We have enough slack on both the rf cables and the rotor cables to take the telescope basically through about 1 and a half rotations and up to 90 degre. Hard limits set for CW rotation: AZ 359, EL 90. Hard limits set for CCW rotation: AZ 315, EL -5. SRTN never commands the rotor to rotate CCW but those limits are set sort of as a runaway telescope catcher.

theDarkLard, 2015-06-19:
Oh and the software doesn't seem to be bugging out with hard limits set, so that's good.

closed 2015-06-25 by theDarkLard