BenningtonCS/Telescope-2014 #115: Webcam

hcrowl, 2015-06-18:
Get the webcam working on the radio telescope computer (using VLC?). If possible, figure out how to integrate the video into srtn.

edaniszewski, 2015-06-22:
My 2 cents: easier to do with VLC (or something similar), for now.

The UI code for srtn is not at all modular. Its actually just inline with all the code that does things throughout a bunch of the files ( #16 )

hcrowl, 2015-06-22:
Agreed. Integration into srtn is probably #dreamland for now. 

VLC (or whatever) should be the first priority.

edaniszewski, 2015-06-24:
Spent some time looking at the best frameworks for creating a more modern (e.g. can actually run on more than just CentOS and Ubuntu). Haven't come to any clear decision yet, but wxWidgets looks neat. It is written in C++, but has bindings in Python. One particularly neat feature relevant to this issue: http://larryo.org/work/information/wxopencv/index.html

theDarkLard, 2015-06-29:
Came up with minimal results during my search today. I have a feeling that either we're using the wrong port, the port isn't open, or that a firewall (?) is blocking the connection to the camera.

hcrowl, 2015-06-29:
Everything seems to work on the camera end. I'm able to reach it on my Mac laptop using VLC.

theDarkLard, 2015-06-30:
Hmm. Yeah then it might be a firewall thing...that was the other main reason people had difficulty getting a stream going. Or it's something else entirely which would not surprise me..ha!

theDarkLard, 2015-07-29:
enabled port 554 (rtsp port for H.264 cameras) in the firewall settings but still getting the same live555/can't connect to mrl error :(

theDarkLard, 2015-07-29:
@hcrowl , when you were setting up the camera the last time, how did you install the browser plug-in that enables you to set up a username and password for your camera? The extension provided by foscam is in an .exe format, while firefox add-ons are in a .xpi format, and chrome add-ons are in a .crx format. Saving the .exe file as either .xpi or .crx corrupts the file.

theDarkLard, 2015-07-29:
And there are converters but all the ones I can find are .crx or .xpi *to* .exe, not the other way around.

hcrowl, 2015-07-29:
I don't think you can view the video stream in a browser (or, at least, I could never get that running), but you should be able to get it running in VLC on one of the Physics lab macs.

theDarkLard, 2015-07-29:
RIght right but this is before that step, this is simply to configure the camera itself and assign to it a username and password. Part of "Step 3: Access the Camera" in the set up manual if that rings a distant bell. You need a browser plug-in to sign in to this window:
[image]

theDarkLard, 2015-07-29:
There is a foscam support page that lists how to force-install the plug-in for all the different browsers, but when you input the url they for each add on, it brings you back to the support page with no file downloaded. This is the page:

http://foscam.us/forum/the-how-to-install-browser-plugins-all-h-264-cameras-t7272.html

hcrowl, 2015-07-29:
That's weird. Maybe try another browser? I think it worked on either Firefox or Chrome without a plugin.

theDarkLard, 2015-07-29:
Yeah I've tried it on both firefox and chrome and both require the plug-in to let you set up your camera. Safari as well. It seems foscam only makes the add-on in a .exe format, or at least that's the only version I can find. Like I said earlier, the instructions to get the other versions of the plug-ins don't bring you to where they are supposed to. Shall I email foscam?

edaniszewski, 2015-07-29:
http://www.foscam.com/Cooperate/FAQinfo/id/45.html

dunno if thats helpful or not

theDarkLard, 2015-07-30:
Works on every linux machine BUT the radio telescope one. I'm not totally sure why or how...I've told the firewall to leave the appropriate ports alone and I've installed the newest version of VLC but still to no avail..but rtsp streaming does in fact work with the webcam.

edaniszewski, 2015-07-30:
While CentOS and Ubuntu are both Linux based, Ubuntu seems to be more
widely-supported than CentOS since it is more popular. Perhaps this is a
CentOS specific problem? Try searching for RSTP streaming issues specific
to CentOS. Perhaps that may provide some enlightenment?

edaniszewski, 2015-07-30:
perhaps a centos issue:
http://w3facility.org/question/centososx-vlc-rtsp-url-does-not-work-on-centos-but-on-osx/

some other things to try:
http://xmodulo.com/live-stream-video-webcam-linux.html
http://www.videolan.org/doc/streaming-howto/en/ch04.html#id311704