BenningtonCS/Telescope-2014 #117: Verify (and document) that HI Line observations work

hcrowl, 2015-06-18:


theDarkLard, 2015-06-29:
Doing this via measuring galactic rotation curve. The galactic center doesn't rise above the horizon until about 10:00 pm so this will have to wait till tonight or tomorrow night.

theDarkLard, 2015-07-03:
I was able to get some help tonight to run the rotation curve survey, it should be finishing up soon!

theDarkLard, 2015-07-03:
A quick peruse through the output file showed vlsr to increase steadily at each point on the galactic plane, so that's promising!

theDarkLard, 2015-07-09:
Running a rotation curve survey~!

theDarkLard, 2015-07-09:
Survey completed. Will take a look at it tomorrow

theDarkLard, 2015-07-14:
Will do this again once pointing is corrected.

theDarkLard, 2015-07-23:
@hcrowl  Some questions:
So here's a plot of a spectrum at galactic longitude 30.
[image]
I have plotted my frequencies in the opposite order MIT has plotted them, so our graph is technically backwards, but my question remains: in the MIT lab write up, they snag the emission at the lowest frequency, so on the end of the slope, not the peak. Is this the correct way to do it? I seem to recall you saying what you want is where that peak shoots up but I could be very mistaken. Just wanted to double check.

hcrowl, 2015-07-23:
It's interesting that it's double-peaked. I wonder if you're seeing the HI near us vs. the HI far from us (so that you see it at two different velocities). 

I think it depends a little bit on what we think the curve should be. At galactic latitude of 30, do we expect the disk to be blueshifted or redshifted? We presumably want to pick the wavelength/frequency that we think is most representative of the disk at this point.

theDarkLard, 2015-07-23:
Right the disk should be more redshifted as we get out from the center. I'm doing another survey now and a little postscript file from srtn confirms.
[image]

theDarkLard, 2015-07-24:
plot
[image]

theDarkLard, 2015-07-28:
I think I'm going to do another survey when the plane is higher up in the sky.