BenningtonCS/Telescope-2014 #121: Correct Pointing Weirdness

theDarkLard, 2015-07-14:
So as of right now, our telescope has a pointing error of about 6.388 (or 5.388) degrees too far to the west. We know our telescope is aligned with true north and our lat/lon values are correct in the software. We know there is some weirdness concerning dates in the time.c file. M1 seems to be nearly impossible to use as a reliable point source because of its proximity to the galactic plane, so it is difficult to deduce a pointing error using it. 

Possible solutions:
-Restow the telescope 6 or 7 degrees east of where it is now.
-Figure out how to correctly update the time.c file (this needs to be done anyway).

theDarkLard, 2015-07-14:
As a sidenote, when you calibrate the telescope and leave it looking at blank sky, the wide bandwidth discrete spectrum does in fact go flat.

theDarkLard, 2015-07-14:
I've been examining the time.c logic and trying to link it with formulas/calculations of GMST that I found online, but some things don't quite match up.
Time.c logic:

````
double gst(double ttime)
{
    double secs, pdum;
    int i;
//    secs = (1999 - 1970) * 31536000.0 + 17.0 * 3600.0 + 16.0 * 60.0 + 20.1948;
    secs = (2011 - 1970) * 31536000.0 + 17.0 * 3600.0 + 15.0 * 60.0 + 58.0778;
    for (i = 1970; i < 2011; i++) {
        if ((i % 4 == 0 && i % 100 != 0) || i % 400 == 0)
            secs += 86400.0;
    }

    return (modf((ttime - secs) / 86164.09053, &pdum) * TWOPI);
/* 17 16 20.1948 UT at 0hr newyear1999 */
/* 17 15 58.0778 UT at 0hr newyear2011 */
}
````

I found a PDF on calculating GMT. It says:

A formula relating the Greenwich mean (not apparent) sidereal time (GST) to the universal time
(UT), good during a given year can be found on page B8 of the Astronomical Almanac. It looks
like this:
GST = G + 0.0657098244 × d + 1.00273791 × t
where d is the day number during the year, t is the UT, and G is a constant for the year, actually
just the GST at UT = 0h on January 0 (= December 31 previous year!).

In the time.c logic, it is calculating GST all in seconds. 2011-1970 is obviously number years since 1970. 17*3600 is I believe the logic for how many seconds in 17 days. Why 17 days? Not sure. The 15 * 60 logic is the same but for minutes. Enitrely unsure what 58.0778 is, perhaps some constant that gets updated every year? I really wish this guy commented his code :/

theDarkLard, 2015-07-14:
the number 58.0778 does not show up in any of my searchings

edaniszewski, 2015-07-15:
I found this http://www2.arnes.si/~gljsentvid10/sidereal.htm which seems to have some useful info in it.This part in particular seems to bring light to the 'random' numbers:

> GMST = 0.48825309298 * 360 = 175.771113474402
>                            = 175.77111 degs
> 
> This corresponds to 11h	39m	5.0672s in time units. ICE gives
> 11h 39m 05.0675s for the same time and date.

so the numbers seem to correspond to GMST (or perhaps GAST?) in time units, which is kinda shown in the comments of the code snippet you posted:

```C
...
/* 17 16 20.1948 UT at 0hr newyear1999 */
/* 17 15 58.0778 UT at 0hr newyear2011 */
...
```

so perhaps the "magic" numbers (17, 15, 58.0778) are the UT time, though I don't really know how they calculated that.. 

In essence, we have:

```
((current_year - 1970) * secs_per_365day) + (17 * secs_per_hour) + (15 * secs_per_min) + 58.0778
```
so if the constants are given by UT time, what this is essentially doing is just converting it all into seconds, giving the seconds since the epoch (1970). 

I think what the code below is doing is accounting for any leap years, adding extra time to account for that.

```C
for (i = 1970; i < 2011; i++) {
        if ((i % 4 == 0 && i % 100 != 0) || i % 400 == 0)
            secs += 86400.0;
}
```

looking on, the `modf` function , described [here](http://www.tutorialspoint.com/c_standard_library/c_function_modf.htm) essentially works such that if
```C
double x, fractpart, intpart;

x = 8.123456;
fractpart = modf(x, &intpart);
```
then
```
intpart = 8.000000
fractpart = 0.123456 
```

so when we have 

```C
return (modf((ttime - secs) / 86164.09053, &pdum) * TWOPI);
```

we are really saying 'take the fraction part of `(ttime - secs) / 86164.09053` and then multiply it by 2pi', and thats our `gst()` return value. Not at all sure where the 86164.09053 is coming from...

theDarkLard, 2015-07-15:
Oh wow totally missed that UT in the comments there. But that's great that means we have everything we need to rewrite that first line, just need to find UT for 0hr for 2015. Yeah the 86164.09053 is a mystery to me as well, although it may be some sort of modified seconds value because there are 86400 seconds in a day. However, *why* this value is shaved in that line escapes me. Thanks a ton for this @edaniszewski !

theDarkLard, 2015-07-15:
Can't find the UT time for 0hr GST for new year 2015 on the internet. The GST calculators that I've found take UTC times instead of UT times so I can't back calculate it. @hcrowl do you have access to an astronomer's almanac or something of the sort that may list these times?

edaniszewski, 2015-07-15:
>  just need to find UT for 0hr for 2015.

we could find it for 0hr for 2015, or we could generate the UT time for the current time (XXhr on XXday for 2015) for even more accuracy. I dont imagine the computational cost of doing this is prohibitive enough to warrant hardcoding of any time value. the less that can be hardcoded, the easier this beast will be to maintain.

theDarkLard, 2015-07-16:
Yeah we also would  never really have to update the time.c file too often. And sorry what I meant by UT for 0hr 2015, I meant UT for 0hr 2015 in GST, if that wasn't already clear.

closed 2015-07-20 by theDarkLard