BenningtonCS/Telescope-2014 #124: Daily Solar Observations

theDarkLard, 2015-07-23:
Hey @hcrowl, I was taking a closer look at MIT's page for daily solar observations, and they observe the sun pretty continuously over a 7 hr period. They also make their observations at 1415.00 MHz, but I have a feeling that that is a pretty arbitrary center freq and they were just trying to get out of the H1 line. But what they do is average the central channels in the spectra together to get a total power measurement, and plot the power measurement over time, over the 7 hr period for each day. Do you think we should follow some version of this procedure or continue to take more brief measurement? The plus side with the long observations is you can catch cool events that you may not otherwise see. Additionally, there will have to be another parser program written to extract the central channels and average them for each spectrum and spit that value back out with the time the spectrum was taken.

edaniszewski, 2015-07-23:
should be fairly easy to create a new script to handle this data, as long as the desired output format is specified.

hcrowl, 2015-07-23:
I agree that the 1415 MHz is probably arbitrary; we would do fine at some slightly higher frequency, I think. Doing longer term observations would be fine/fun, and might be an interesting longer-term project. 

I agree with both of you that it should be straightforward to write code to handle multiple observations per day and to average frequencies together. 

More challenging would be to figure out our calibration strategy. How often do we need to calibrate? Once a day? Once a week? How do changing weather conditions change our calibration?

theDarkLard, 2015-07-24:
We could do the advanced calibration daily and, saving the data for Phot, Pcold, Thot, see how the changes in these values affect the changes in Tsys. Once we get an idea of a sort of timescale that Tsys changes over, and with making special note of the weather each day, I think we could come up with a general strategy.