BenningtonCS/Telescope-2014 #125: Calibration Scheme/Characterize roof noise

theDarkLard, 2015-07-28:
Plots of roof noise vs. degrees of elevation at different azimuthal points:
[image]
[image]

theDarkLard, 2015-07-28:
210 and 330 were points right in the middle of the tallest, broadest peaks in this plot:
[image]
90 was more of a control that shows a curve quite similar to the noise levels at 0 degrees azimuth.

theDarkLard, 2015-07-28:
I think the radio tower in north bennington is a good explanation for the high levels of noise in the 300-360 range, but the noise in the 210 range is coming from somewhere who's radio capabilities are unknown. It's way out the way of Mt. Anthony, I just went up and checked.

theDarkLard, 2015-07-28:
Also the Tsys for 1420.406 MHz was 10 degrees less today than it was the last couple times we did the advanced calibration.

theDarkLard, 2015-07-28:
I think I'll do these scans again at the same places a little later this afternoon to see how time dependent these levels are.

theDarkLard, 2015-07-28:
The Az 90 scan was done at 3pm yesterday, the 210 scan done at 9:45 am this morning, the 330 scan done at 10:10 am

theDarkLard, 2015-07-28:
@hcrowl deduced that it was the trees surrounding dickinson that were causing all of the interference on the horizon. Darn biology always mucking up our astronomy!
Anyway, break down is like this:
0-200 deg az  -- make observations above 15 degrees el
200 - 260 -- make observations above 30 degrees el
260 - 290 -- make observations above 15 degrees el
290 - 350 -- make observastions above 25 degrees

theDarkLard, 2015-07-28:
calibration scheme bit related to #116

edaniszewski, 2015-07-28:
> Anyway, break down is like this:
> 0-200 deg az -- make observations above 15 degrees el
> 200 - 260 -- make observations above 30 degrees el
> 260 - 290 -- make observations above 15 degrees el
> 290 - 350 -- make observastions above 25 degrees el

@theDarkLard this could be good to capture in the wiki somewhere, so one won't have to dig through all the issues to find it.

hcrowl, 2015-07-28:
>>Anyway, break down is like this:
>>0-200 deg az -- make observations above 15 degrees el
>>200 - 260 -- make observations above 30 degrees el
>>260 - 290 -- make observations above 15 degrees el
>>290 - 350 -- make observastions above 25 degrees el

>@theDarkLard this could be good to capture in the wiki somewhere, so one won't have to dig through all the issues to find it.

Agreed. Perhaps with a caveat that these are conservative limits and conditions are likely to be much more favorable at these low altitudes in the winter time?

theDarkLard, 2015-07-28:
sounds good, a wiki page is being constructed.

theDarkLard, 2015-07-28:
wiki page done.

theDarkLard, 2015-07-28:
I did another 'advanced calibration' at 1420.4 MHz, Tsys is up to 188.66 K (at 2:30 pm) from 173.8 K this morning at 9 am. Don't have enough data to verify this, but it seems like Tsys might fluctuate a lot with the ambient temp.

theDarkLard, 2015-07-28:
#Date           Phot            Pcold           Thot            Tsys            Freq            Weather/misc.

7/27/15         484.3           190.5           195.9           187.2           1420.406        Heavy cloud cover

7/27/15         471.3           174             302             171.9           1421.5          Clear

7/28/15         458.6           172             298.2           173.8           1420.406        Clear, morning

7/28/15         469.8           182             301.5           185.9           1421.5          Clear

7/28/15         459             179             302             188.66          1420.406        Clear, afternoon

this is all the data we have right now for those who are curious

theDarkLard, 2015-07-29:
@hcrowl is there anyway we could get some batteries for the portable vernier interafces? I've been taking the ambient temperarature from the weather sation on campus as Thot because the verniers keep running out of battery, but I compared the temp on the roof with the temp from the weather station, and the roof was 2.5 degrees (kelvin) warmer than the temp reported by the weather station (probably because of the black roof). That difference has almost certainly been messing wiht our Tsys values so the data I've collected the last few days may be bunk. I am going to subtract off 2.5 degrees from Thot and recalculate to see how those values compare to today's.

theDarkLard, 2015-07-29:
*Add

theDarkLard, 2015-07-29:
As expected the recalculation just bumped up all the Tsys values by a couple degrees, I guess not enough to be too concerned about, but a working temperature probe would still be very useful.