BenningtonCS/Telescope-2014 #126: add test data to repo

edaniszewski, 2015-08-06:
@theDarkLard 

would it be possible to add a directory to the repo that contains various .rad data files that you've generated over the summer?

the only one I've had available to work with/test against is the one that was originally checked into the repo long ago. It would be nice to have a bunch of different files to test against.

could have the directory structure look something like:

```
Telescope-2014 /
  | - documentation /
  | - src /
  | - utils /
  | - data /
       | - solar_obsertvations /
             | - solar_observation_1.rad
             | - solar_observation_2.rad
             | - ...
       | - galactic_plane_observations /
             | - galactic_observation_1.rad
             | - galactic_observation_2.rad
             | - ...
       | - ...
  | - .gitignore
```

or however it seems best to group them

theDarkLard, 2015-08-06:
> to @theDarkLard <https://github.com/theDarkLard>.

Yeah I'll dump all the important ones I've collected so far. How do you
commit stuff to the repo....?

edaniszewski, 2015-08-06:
If you don't feel comfortable doing it with the command line, there are a bunch of GUI tools you can use: https://git-scm.com/download/gui/linux

you would just need to point it to the existing repo. Then add your stuff to wherever you are putting it inside of the local copy of your repo (e.g. somewhere in the Telescope-2014 folder on your machine), then there should be an option on the GUI to commit + push.

If you wanna try out with the command line, you should just add the files somewhere in the local copy of your repo (same as above), then `cd` into the repository, e.g.
```
$ cd ~/Documents/Repositories/Telescope-2014
```
or wherever it is on your local machine.
then, doing `git status` should show something like

```
$ git status
On branch master
Your branch is up-to-date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)

        data/solar_observations/observation1.rad
        data/solar_observations/observation1.rad

nothing added to commit but untracked files present (use "git add" to track)
```
which should list all the files you added. do a `git add` for each of them

```
$ git add data/solar_observations/observation1.rad
...
```
commit the changes

```
$ git commit -m "add data files"
```

and push it to github

```
git push
```

I think that should basically be it.. feel free to lemme know if ya need help with it/if I goofed and forgot a step somewhere or something.

closed 2015-08-08 by edaniszewski